OpenXiangShan/difftest #1: difftest running error

wangpengabc, 2021-07-23:
```
$:~/riscv/NutShell/difftest$ ./build/emu -b 0 -e 0 -i ./ready-to-run/linux.bin --diff ./ready-to-run/riscv64-nemu-interpreter-so

```
I pulled NutShell and difftest rep to my local envierment,  NutShell switched to difftest branch. After running "make emu" in difftest dirctory, i run the command like above. Terminal promopts : 
```
Emu compiled at Jul 23 2021, 16:27:44
The image is ./ready-to-run/linux.bin
Using simulated 8192MB RAM
[warning] sdcard img not found
Using ./ready-to-run/riscv64-nemu-interpreter-so for difftest
emu: /home/wangpeng/riscv/NutShell/difftest/src/test/csrc/difftest/nemuproxy.cpp:48: NemuProxy::NemuProxy(int): Assertion `memcpy_from_ref' failed.
Aborted (core dumped)
```

AugustusWillisWang, 2021-07-23:
NutShell difftest branch does not use pre-compiled NEMU in `ready-to-run`. Build NEMU (riscv64-nemu-interpreter-so) manually using NEMU [here](https://github.com/OpenXiangShan/xs-env), and try again.

Your can refer to [this module](https://github.com/OpenXiangShan/xs-env) and [this script](https://github.com/OpenXiangShan/xs-env/blob/master/setup.sh) for further instructions.

wangpengabc, 2021-07-23:
> NutShell difftest branch does not use pre-compiled NEMU in `ready-to-run`. Build NEMU (riscv64-nemu-interpreter-so) manually using NEMU [here](https://github.com/OpenXiangShan/xs-env), and try [...]

I build NEMU manually from NEMU rep, the result did not change:
```
~/riscv/NutShell$ ./build/emu -b 0 -e 0 -i ./ready-to-run/microbench.bin --diff /home/wangpeng/riscv/NutShell_env/nemu/build/riscv64-nemu-interpreter-so
Emu compiled at Jul 23 2021, 16:27:44
The image is ./ready-to-run/microbench.bin
Using simulated 8192MB RAM
[warning] sdcard img not found
Using /home/wangpeng/riscv/NutShell_env/nemu/build/riscv64-nemu-interpreter-so for difftest
emu: /home/wangpeng/riscv/NutShell/difftest/src/test/csrc/difftest/nemuproxy.cpp:48: NemuProxy::NemuProxy(int): Assertion `memcpy_from_ref' failed.
Aborted (core dumped)
```

AugustusWillisWang, 2021-07-23:
Can you show me your NEMU's commit id? Which branch are you on?

wangpengabc, 2021-07-23:
> Can you show me your NEMU's commit id? Which branch are you on?
```
:~/riscv/NutShell_env/nemu$ git log
commit 85edc31846b3813adab4f07779a1f74ba54a3d2c (HEAD -> master, origin/master, origin/HEAD)
Merge: dc7c4d3 92e18d7
Author: wakafa <wangkaifan@ict.ac.cn>
Date:   Thu Jul 30 05:18:17 2020 -0500

    Merge pull request #2 from OSCPU/fix-updateAD
    
    riscv64,mmu: avoid updating PTE.A/D in so-file

```

wangpengabc, 2021-07-23:
> Can you show me your NEMU's commit id? Which branch are you on?

When i change to this commit version, it works! Thanks 
```
:~/riscv/NutShell_env/xs-env/NEMU$ git log
commit 145db2d924ef4a39a3ea69b95e0392abcf282646 (HEAD, origin/xs, origin/HEAD, xs)
Author: William Wang <zeweiwang@outlook.com>
Date:   Wed Jul 14 20:41:15 2021 +0800

    rv64: disable store difftest when sq overflow
    
    It is a temp patch, store commit queue check will be refactored later
```

closed 2021-07-23 by wangpengabc