OpenXiangShan/difftest #86: Difftest中的波形功能应该如何开启？

Pokusala, 2022-09-27:
我使用了Difftest作为riscv处理器的代码比对工具，修改了difftest/config/verilator.mk中关于波形的配置，设置EMU_TRACE为1，并且在执行时添加了--dump-wave参数。执行参数如下：
`./build/emu -b 0 -e 10000 -i ./ready-to-run/rtthread.bin --dump-wave`
在执行过程中，控制台显示
`dump wave to /home/xx/rvcpu/1/rvcpu/build/2022-09-27@04:21:31.vcd...`
在这个目录下，我找到了这个波形文件，使用gtkwave打开后，所有的信号名均能正常显示，但是波形图显示只有一个时间点而不是完整的执行周期，并且为difftest执行最后一刻的值。我想咨询一下这是否因为我的使用方法有问题，正确的显示波形应该如何设置。

poemonsense, 2022-10-12:
参考：https://github.com/OpenXiangShan/difftest/blob/master/src/test/csrc/verilator/emu.cpp#L299-L303

可以看到，波形的周期数是依据
```
 auto trap = difftest[0]->get_trap_event();
    uint64_t cycle = trap->cycleCnt;
```
得到的

如果没有接difftest，需要把这里改成自己算的一个周期数。如果接了difftest，可以检查一下周期是否是正确的

Pokusala, 2022-10-13:
> 参考：https://github.com/OpenXiangShan/difftest/blob/master/src/test/csrc/verilator/emu.cpp#L299-L303
> 
> 可以看到，波形的周期数是依据
> 
> ```
>  auto trap = difftest[0]->get_trap_event();
>     uint64_t cycle = trap->cycleCnt;
> ```
> 
> 得到的
> 
> 如果没有接difftest，需要把这里改成自己算的一个周期数。如果接了difftest，可以检查一下周期是否是正确的

好的，是我之前没有给difftest的cycleCnt连线导致的，感谢您的解答。

closed 2022-10-13 by Pokusala