OpenXiangShan/difftest #117: Snapshot timestamp t not updated in while loop for simulation

shili2017, 2023-03-25:
Timestamp t is initialized at [this line](https://github.com/OpenXiangShan/difftest/blob/master/src/test/csrc/verilator/emu.cpp#L364). In the while loop for simulation, t is used at [this line](https://github.com/OpenXiangShan/difftest/blob/master/src/test/csrc/verilator/emu.cpp#L470) for generating snapshot, but is never updated in the while loop, so no snapshot is actually generated.

shili2017, 2023-03-25:
It seems that this feature is not maintained any more?

wakafa1, 2023-07-24:
> It seems that this feature is not maintained any more?

Yes. Since snapshot can only save the state of verilator, it is difficult to save the state of other Difftest components (e.g., nemu), we don't use snapshot much anymore but use lightsss instead.

closed 2023-07-24 by shili2017