OpenXiangShan/difftest #220: When using Spike as REF, it fails to correctly synchronize the DUT's state during exceptions

youzi27, 2023-12-01:
When using Spike as the REF and conducting difftest with the rocket-chip, I noticed that REF records an incorrect value for _mepc_ during exceptions. 

The Commit Instr Trace is as follows：

```
- [468] commit pc 00000000800006ec inst 6141606f wen 1 dst 00 data 0000000000000000 idx 000 j       pc + 0x16614
- [469] commit pc 0000000080016d00 inst 00000013 wen 1 dst 00 data 0000000000000000 idx 000 nop
- [470] exception pc 0000000080016d04 inst 00000000 cause 0000000000000002 c.unimp//EXCEPTION
- [471] commit pc 0000000080002000 inst 0400006f wen 1 dst 00 data 0000000000000000 idx 000 j       pc + 0x40
- ...
- [502] commit pc 00000000800020b8 inst 0dbc3c23 wen 0 dst 24 data 0000000080001fe0 idx 000 sd      s11, 216(s8)
- [503] commit pc 00000000800020bc inst 0fcc3023 wen 0 dst 00 data 0000000000000000 idx 000 sd      t3, 224(s8) <--
- [504] commit pc 00000000800020d0 inst ffffe297 wen 1 dst 05 data 00000000800000d0 idx 000 auipc   t0, 0xffffe
- ...
- [508] commit pc 0000000080016d00 inst 00000013 wen 1 dst 00 data 0000000000000000 idx 000 nop
```

**Difftest result：**

```
- pc: 0x0000000080002000 mstatus: 0x0000000a00001800 mcause: 0x0000000000000002 mepc: 0x00000000800020bc
-                        sstatus: 0x0000000200000000 scause: 0xaa75e257df5869e5 sepc: 0x0000006f10958978
- satp: 0x0000000000000000
- mip: 0x0000000000000220 mie: 0x0000000000000000 mscratch: 0x6410e61c5e18874a sscratch: 0x6410e61c5e18874a
- mideleg: 0x0000000000000000 medeleg: 0x0000000000000000
- mtval: 0x0000000000000000 stval: 0xfffffff9ba1e5c49 mtvec: 0x0000000080002001 stvec: 0x00000000314fc1c8
- privilege mode:3
- priviledgeMode: 3
-    mepc different at pc = 0x00800020bc, right= 0x00000000800020bc, wrong = 0x0000000080016d04
- Core 0: ABORT at pc = 0x800020c0
```

The above log found that ref gave the wrong _mepc_. The correct value should be: _0x0000000080016d04_

To verify whether it's an error within Spike itself, I ran it independently in Spike, and the debug log is as follows:

```
- core   0: 0x00000000800006ec (0x6141606f) j       pc + 0x16614
- (spike) 
- core   0: >>>>  _real_end
- core   0: 0x0000000080016d00 (0x00000013) nop
- (spike) 
- core   0: >>>>  _end
- core   0: 0x0000000080016d04 (0x00000000) c.unimp
- core   0: exception trap_illegal_instruction, epc 0x0000000080016d04
- core   0:           tval 0x0000000000000000
- (spike) 
- core   0: >>>>  mtvec_handler
- core   0: 0x0000000080002000 (0x0400006f) j       pc + 0x40
- (spike) reg 0 mepc
- 0x0000000080016d04
```

I found that Spike itself is correct, so the issue might be with the synchronization mechanism of difftest when it encounters exceptions.

Img and log files: 
[test.zip](https://github.com/OpenXiangShan/difftest/files/13526208/test.zip)

youzi27, 2023-12-01:
The above situation seems to occur during nested exceptions. I tested a standalone illegal instruction exception, and it worked fine.

```
- [446] commit pc 00000000800006f8 inst 2110206f wen 1 dst 00 data 0000000000000000 idx 000 j       pc + 0x2a10
- [447] commit pc 0000000080003108 inst 00000013 wen 1 dst 00 data 0000000000000000 idx 000 nop
- [448] exception pc 000000008000310c inst 00000000 cause 0000000000000002 c.unimp
- [449] commit pc 0000000080002000 inst 0400006f wen 1 dst 00 data 0000000000000000 idx 000 j       pc + 0x40
- [450] commit pc 0000000080002040 inst ff878793 wen 1 dst 15 data 0000000080016cf0 idx 000 addi    a5, a5, -8
```

```
- pc: 0x0000000080016d04 mstatus: 0x0000000a00001800 mcause: 0x0000000000000002 mepc: 0x000000008000310c
-                        sstatus: 0x0000000200000000 scause: 0xaa75e257df5869e5 sepc: 0x0000006f10958978
- satp: 0x0000000000000000
- mip: 0x0000000000000220 mie: 0x0000000000000000 mscratch: 0x6410e61c5e18874a sscratch: 0x6410e61c5e18874a
- mideleg: 0x0000000000000000 medeleg: 0x0000000000000000
- mtval: 0x0000000000000000 stval: 0xfffffff9ba1e5c49 mtvec: 0x0000000080002001 stvec: 0x00000000314fc1c8
- privilege mode:3
- priviledgeMode: 3
- Core 0: ABORT at pc = 0x80016d00
```
The mepc matches.

poemonsense, 2023-12-06:
Difftest does not sync registers upon exceptions. What it does it to **compare the states of REF with the states of DUT**. `mepc` of Spike is never affected by the DUT.

From your logs, Spike raises an illegal instruction exception on 0x00000000800020bc with mtval of zero. Please check your workloads and Spike logs to see why Spike sees this.

closed 2023-12-06 by poemonsense