OpenXiangShan/difftest #226: AbstractRefProxy 声明了与reference model 不相同的函数原型

gxflying, 2023-12-20:
AbstractRefProxy 的函数声明在这里，其中以ref_regcpy为例，从宏定义可知，函数原型包含3个参数
[ f(ref_regcpy, difftest_regcpy, void, void*, bool, bool)](https://github.com/OpenXiangShan/difftest/blob/cbcc801699c7b20de84f3bc15f03f3f24c72d400/src/test/csrc/difftest/refproxy.h#L77)

但是在NUMU中的 [void difftest_regcpy(void *dut, bool direction, bool restore, uint64_t restore_count)](https://github.com/OpenXiangShan/NEMU/blob/8ce7caa53052427c47fb215c0736a75a4fb0d53a/src/cpu/difftest/ref.c#L101) 原型是有4个参数？ 

这里是不是有点问题，还请帮忙澄清一下

https://github.com/OpenXiangShan/difftest/blob/cbcc801699c7b20de84f3bc15f03f3f24c72d400/src/test/csrc/difftest/refproxy.h#L77

poemonsense, 2023-12-21:
目前Spike实现了第四个参数`on_demand`（参考：https://github.com/OpenXiangShan/riscv-isa-sim/blob/difftest/difftest/difftest.cc#L427 ）

NEMU目前还没有实现。因为不实现这个参数，在大多数情况下不会影响功能正确性，所以一直没有对NEMU添加支持

如果感兴趣，欢迎给NEMU添加这个参数

closed 2023-12-22 by poemonsense

gxflying, 2023-12-22:
参数不同生成的函数调用的栈帧就不同，特别是在实现参数少，传入参数多的情况下，容易引起数据被覆盖，即使目前没问题，可以考虑添加到todo list，或备忘录

poemonsense, 2023-12-22:
欢迎给NEMU提PR