OpenXiangShan/difftest #248: 覆盖率导向模糊测试相关流程使用

wyz-icer, 2024-01-15:


wyz-icer, 2024-01-15:
2023 年 8 月 24 日@第三届 RISC -V 中国峰会 内公布内容，香山当前流程支持   " CPU 的覆盖率导向模糊测试" 相关内容
结合当前difftest 提交版本  master 分支内 scripts/coverage 
相关脚本分析，主要支持功能：
1.  依verilator 识别规则对verilog 文件进行预处理 以 统计需要范围内的覆盖情况
2.  获取代码覆盖结果数据

问题： 
1. 有没有更多的关于覆盖率导向模糊测试的 流程说明 或者使用说明； 
2. 在哪儿可以看到 峰会提到的 覆盖率收集闭环流程： 收集--算法推导分析代码覆盖率--测试激励再输入--协同仿真 的流程迭代。

poemonsense, 2024-01-15:
Please refer to https://github.com/OpenXiangShan/xfuzz

closed 2024-01-15 by poemonsense

wyz-icer, 2024-01-16:
好的 感谢！

poemonsense, 2024-01-16:
> > Please refer to https://github.com/OpenXiangShan/xfuzz
> 
> 关于这个仓库有没有更详细一点的使用说明

目前没有了。这个仓库的所有命令都已经列在README里面了

wyz-icer, 2024-01-16:
> > > Please refer to https://github.com/OpenXiangShan/xfuzz
> > 
> > 
> > 关于这个仓库有没有更详细一点的使用说明
> 
> 目前没有了。这个仓库的所有命令都已经列在README里面了

比如readme 内提到的 集成design内 提到的几点
1. it could be passed to [Verilator](https://github.com/verilator/verilator) as an external library for linking.  //如何配置link至verilator
2. The required interfaces between Rust and C/C++ modules are listed exclusively at [the harness file] (https://github.com/OpenXiangShan/xfuzz/blob/main/src/harness.rs).  //这些接口是和difftest内交互用的吗？ 
3. have upgraded the [DiffTest](https://github.com/OpenXiangShan/difftest) environment to support these interfaces.  //如何配置同difftest交互
4. Please refer to the [DiffTest README.md](https://github.com/OpenXiangShan/difftest/blob/master/README.md) for connecting your CPU design with the fuzzer.  //difftest 支持生成verilog相关接口命令”make difftest_verilog NOOP_HOME=$(pwd)“  ， 如何同xfuzz进行连接？

poemonsense, 2024-01-16:
可以参考 CI 流程中的 Difftest with Coverage-Guided Fuzzer 部分：https://github.com/OpenXiangShan/difftest/blob/master/.github/workflows/main.yml