OpenXiangShan/difftest #332: move指令消除问题

Jeacy, 2024-04-07:
请问difftest中为什么会对move指令跳过比对？无论是否消除move在提交后一拍的逻辑寄存器应该是一致的

poemonsense, 2024-04-08:
从哪里得到的【对move指令跳过比对】这一结论？

Jeacy, 2024-04-08:
抱歉，我应该是理解错误了。以下代码的意思应该是先判断是否为move消除后，再根据是否为mmio和perfcnt来决定跳过比对。我想问的应该是，为什么通过move消除来判断是否skip？
[image]

poemonsense, 2024-04-08:
Git blame可以看到这一行代码的修改历史。上一次修改的commit是https://github.com/OpenXiangShan/XiangShan/commit/7d45a146d3c44839ba821bb91ca4950dc2b817f2

看到修改的代码中有代码注释解释您的这一问题，可能是当时修改时被误删了：

>       // when committing an eliminated move instruction,
>       // we must make sure that skip is properly set to false (output from EXU is random value)
> 

另外，这个问题是香山设计的问题，和difftest无关，我就先close这个issue了。为了跟踪问题和方便其他的用户查询，如果您对这个问题后续仍然有疑问，烦请移步香山仓库提issue询问：https://github.com/OpenXiangShan/XiangShan/issues

closed 2024-04-08 by poemonsense

Jeacy, 2024-04-08:
好的，感谢您的回答。