OpenXiangShan/difftest #374: 时钟及rst配置问题

wyz-icer, 2024-05-11:


wyz-icer, 2024-05-11:
看着波形里dump的是rst从1 开始的，时钟周期是2ps， 麻烦问下 这两个在哪里配置的

poemonsense, 2024-05-12:
C++ code for dumping wavefrom is at https://github.com/OpenXiangShan/difftest/blob/master/src/test/csrc/verilator/emu.cpp#L610-L618

1) By default, `--dump-wave` will dump the waveform only at clock=1 after the reset is deasserted. This option greatly (50%) reduces the waveform size and the simulation time. Using `--dump-wave-full` will dump the full waveform at the cost of lower simulation speed.

2) We are using the default timescale option from Verilator (see https://veripool.org/guide/latest/exe_verilator.html?highlight=timescale#cmdoption-timescale), which is 1ps/1ps. It's worth noting that, for RTL simulation, timescale does not actually affect the simulation results. Commercial RTL simulators are using the timescale to match real time. However, this is meaningless for early-stage RTL simulation.

(Question issues are by default closed when answered. Please feel free to leave future comments in this issue for further discussions.)

closed 2024-05-12 by poemonsense