OpenXiangShan/difftest #395: Proposal to remove ssh remote on verilator.mk

cyyself, 2024-06-13:
In `verilator.mk`, we have a feature that SSH to the remote host to build if we have a `$REMOTE` set. However, this feature is not well maintained.

For example, we didn't pass the `RTL_SUFFIX` variable to the remote host, and it will use `.v` by default, but currently, NutShell generated RTL with `.sv,` so this feature does not work. Similar problems are also caused by `WITH_CHISELDB` and `WITH_CONSTANTIN` for NetShell.

Although it can be simply fixed by adding these parameters. However, some features are easy to forget and cause some problems. For example, we currently have PGO_EMU_ARGS, which may have space in the string. If we use some command like `ssh $(REMOTE) make PGO_EMU_ARGS=$(PGO_EMU_ARGS)`, the space will be parsed as the splitter for each argument. I don't know how we need this feature today since the compile time is mostly bounded to single-thread performance since verialtor generates C code takes time. A server with more core helps, but it is not critical.

Tang-Haojin, 2024-06-13:
It is suggested not to remove this feature for now. If the parallel compilation phase of C++ is run on a server with fewer cores, it will take relatively longer time, and it will also have a greater impact on the development of others compared to the servers with more cores. If you feel that the maintenance cost is too high, you need not to support this feature on PGO.

cyyself, 2024-06-13:
The problem is addressed in #396. I think we don't need to remove ssh remote now.

closed 2024-06-13 by cyyself