OpenXiangShan/difftest #436: 仿真加速

changekkk, 2024-08-09:
你好，关于仿真加速器和FPGA环境上的difftest目前已经开发完毕了吗？是否有开源可用的环境

klin02, 2024-08-10:
目前已经实现的仿真加速功能已经全部开源至本仓库，支持单核/双核加速，并已经在仿真加速器帕拉丁上进行使用。FPGA及进一步加速功能仍在开发当中。

At present, all realized simulation acceleration features have been open source in this repo, supporting single-core/dual-core Verification. Difftest have been used on the emulator Palladium. 

Difftest on FPGA and further acceleration features are still WIP.

changekkk, 2024-08-11:
你好，这里提到已实现的加速功能对应的的difftest版本是哪一个

klin02, 2024-08-11:
已经实现的加速功能都已经合入 master 分支，通过 chisel 生成 verilog 时传入对应的 difftest-config 开启。当前支持的加速功能包含非阻塞传输及比对、数据语义压缩及 DPIC 函数合并传输等。

All implemented acceleration features have been merged into master branch, which can be turned on by passing the corresponding difftest-config when chisel generates verilog. Currently supported acceleration features include non-blocking transfer and compare, data semantic squash, and DPIC function batching.

changekkk, 2024-08-15:
你好，想问一下，前面提到的在帕拉丁使用的difftest版本，与FPGA及进一步加速功能是一类的吗，帕拉丁本身也是FPGA设备，对于difftest来说，在帕拉丁跑和FPGA上跑是不同版本吗？

klin02, 2024-08-15:
目前的加速功能硬件实现部分是通用的，主要在通信方式有所区别。帕拉丁支持直接使用 DPIC 函数、软硬件协同仿真。FPGA 预计将使用 PCIe 进行实现，相关适配尚未完成。

changekkk, 2024-08-16:
好的，咱们Difftest在FPGA上使用的话，主要是如何跟fpga flow配合调试的？在FPGA上跑设计代码，然后同步difftest运行nemu来比对吗

klin02, 2024-08-17:
FPGA流程与仿真加速器类似，Difftest 将负责：硬件侧从待测设计提取信号，通过PCIe进行传输；软件侧接收数据并处理，驱动nemu进行比对

changekkk, 2024-08-18:
这里FPGA流程预计会比仿真加速器快多少，另外FPGA流程的话，是FPGA运行DUT，然后Difftest主要来提取运行结果并通过PCIe来驱动nemu比对吗？

klin02, 2024-08-18:
整体流程基本上是这样的。目前 FPGA 流程尚未跑通，尚无准确的数据，根据部分相关工作的速度数据及数据量对比，预估速度会达到数至数十 MCycle/s。

changekkk, 2024-08-19:
好的，谢谢

changekkk, 2024-10-22:
您好，目前基于FPGA使用的Difftest环境，有具体的使用流程和对应环境版本介绍吗？

changekkk, 2024-10-24:
> FPGA流程与仿真加速器类似，Difftest 将负责：硬件侧从待测设计提取信号，通过PCIe进行传输；软件侧接收数据并处理，驱动nemu进行比对

您好，还想问一下，咱们基于FPGA的Difftest flow，具备硬件的snapshot功能吗？还是说仅支持硬件级的比对

klin02, 2024-10-28:
目前 FPGA difftest 还没有完全跑通，跑通之后会整理对应的使用文档。

目前规划的 FPGA flow 主要使用 FPGA 作为 DUT 的仿真，比对数据将采集后通过 PCIe 传输至软件侧进行比对。考虑到香山的规模以及资源限制，应该不会直接进行硬件级电路 Snapshot，而是在 FPGA 进行错误定位后，由 Palladium 进行错误位置的细粒度重现。

closed 2025-01-10 by klin02