OpenXiangShan/difftest #458: 替换设计verilator编译错误

wyz-icer, 2024-09-04:


wyz-icer, 2024-09-04:
基于difftest 仿真框架在替换设计过程中编译报错，
调整将原 rocketchip chisel 生成的 verilog 设计文件替换为 目标设计，
通过 -f filelist  SimTop.v 指定设计范围，
verilator 编译过程报错。

poemonsense, 2024-09-04:
Thank you for reporting.

We are working on improve the process of bug report toward more efficient debugging from us for DiffTest users. Please use the following template to report the bug. Some necessary information in the template is required to help us address the issue.

```
<!-- Please use ENGLISH words to describe the issues. -->
<!-- Using English helps the knowledge spread across the community. -->
<!-- If you don't know English, please use Google Translate or ChatGPT. -->
<!-- We reserve the rights to edit or delete non-English comments. -->

<!-- Choose at least ONE of the DiffTest components -->
<!-- Note: we may only respond to issues related to DiffTest. -->
**Related component**: build | simulation framework | cosim with a REF

**Describe the bug**
A clear and concise description of what the bug is.

<!-- Please list ALL changes you have made to the source code if applicable -->
<!-- Note: if we cannot reproduce, we are probably unable to help you. -->
**To Reproduce**
Steps to reproduce the behavior:
1. Clone the repository
2. Edit '....'
3. Build with the command '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

<!-- describe your thoughts on the issue -->
<!-- Note: you and us working together helps the debugging process -->
**What you expect us (DiffTest developers) to do for you**

**Additional context**
Add any other context about the problem here.
```

closed 2024-09-08 by wyz-icer