OpenXiangShan/difftest #503: What happened when three different ends, and why.

ha0lyu, 2024-11-13:
**Related component**: cosim with a REF

**Describe the question**
I built the env by xs-env and run emu to test xiangshan, the ref model is NEMU. Recently, I encountered 3 different ends I have never met.
1. When instr trace & ref regs are printed:
1.1 "Core 0 dump: critical_error raise" printed.
       **What happened to here? Why this happened?**
1.2 "Core 0: [31mABORT at pc = "
       Usually, when "Core 0: [31mABORT at pc = " printed, it will print "xxx different at pc =", but this time, it **did not** print it.
       This is abnormal, isn't it?

2. “Assertion failed at /home/xs-env/XiangShan/build/rtl/DCache.sv:1430. The simulation stopped. There might be some assertion failed.”:
      **What happened and why?** Is there a bug triggered it?

**What you expect us (DiffTest developers) to do for you**
I need someone can help me to figure out those questions.

**Thanks a lot.**

poemonsense, 2024-11-13:
> 1. When instr trace & ref regs are printed:
>    1.1 "Core 0 dump: critical_error raise" printed.
>    **What happened to here? Why this happened?**

See https://github.com/OpenXiangShan/difftest/pull/486

>    1.2 "Core 0: [31mABORT at pc = "
>    Usually, when "Core 0: [31mABORT at pc = " printed, it will print "xxx different at pc =", but this time, it **did not** print it.
>    This is abnormal, isn't it?

Please list full messages for us to help you.

> 2. “Assertion failed at /home/xs-env/XiangShan/build/rtl/DCache.sv:1430. The simulation stopped. There might be some assertion failed.”:
>    **What happened and why?** Is there a bug triggered it?

Simply as it says, the assertion fails.

ha0lyu, 2024-11-13:
Here is the tail of a log:
```
......
[24] commit pc 0000000080000014 inst 00000297 wen 1 dst 05 data 0000000080000014 idx 00e auipc   t0, 0x0
[25] commit pc 0000000080000018 inst 00028293 wen 1 dst 05 data 0000000080000014 idx 00f mv      t0, t0
[26] commit pc 000000008000001c inst 30529073 wen 0 dst 00 data 0000000000000000 idx 010 csrw    mtvec, t0 <--
......
virtualization mode: 0
privilege mode:3
pmp: 16 entries active, details:
......
vtype: 0x8000000000000000 vstart: 0x0000000000000000 vxsat: 0x0000000000000001
vxrm: 0x0000000000000003 vl: 0x0000000000000000 vcsr: 0x0000000000000007
tselect: 0x0000000000000000
privilegeMode: 3
Core 0: [31mABORT at pc = 0x80000e78
[0m[35mCore-0 instrCnt = 7,025, cycleCnt = 70,921, IPC = 0.099054
[0m[34mSeed=0 Guest cycle spent: 70,924 (this will be different from cycleCnt if emu loads a snapshot)
[0m[34mHost time spent: 126,463ms
[0m
```
I guess there is a mismatch of `mtval` reg, but there was no message about it, especially `different at pc =`. I ran this workload again, it shows. It may be a problem with the previous redirection of output, but now it's shows in terminal.

Thank you sir.

poemonsense, 2024-11-13:
Please list the *full* message here. Otherwise, we are unable to help.

ha0lyu, 2024-11-18:
Please note the "In the last commit group" field in log files. Based on A, mentioned above, I test B, C & D.

**B**: Do something before `nop` at `raaddr` function:
```
raaddr:
    # DO SOMTHING #
    nop
    nop
```
B-1: `addi t0, zero, 0x50`. XiangShan will execute it bug NEMU won't.
B-2:  `addi t0, zero, 0x0` and `la t0, raaddr`: NEMU did not load the address, XiangShan load a wrong value.
[case-B.zip](https://github.com/user-attachments/files/17801484/case-B.zip)

ha0lyu, 2024-11-18:
Last one D: XiangShan load `raaddr` correctly:
```
raaddr:
    la t0, raaddr
    nop
    nop
```
we got: 
> t0 different at pc = 0x0080000010, right= 0x0000000000000000, wrong = 0x0000000080000010

[C.log](https://github.com/user-attachments/files/17801641/C.log)
[D.log](https://github.com/user-attachments/files/17801642/D.log)

**Environment**
  XiangShan branch: Master
  XiangShan commit id: [011f1ef](https://github.com/OpenXiangShan/XiangShan/commit/011f1effdbdc7c6fa84e1de650ccc95e54e360b6)
  NEMU commit id: [b9507fb](https://github.com/OpenXiangShan/NEMU/commit/b9507fbc62ef4f2acb0bfc1e4e9f495e075aaa40)

ha0lyu, 2024-11-18:
I am not sure whether it is bugs of XiangShan or NEMU. Because I got the **1.2 ABORT** first, and I checked it got those results.

poemonsense, 2024-11-18:
For us as the cosim framework developers, we are responsible for ensuring DiffTest reports the correct error information to the user. We will not respond to any bugs you encountered on any designs, because that is out of the scope of this repo.

Thus, if you are facing DiffTest errors on XiangShan with some error messages, please raise an issue in the XiangShan repo.

If you are facing a case where difftest aborts but no error information is shown, please list the _full_ message here, so that we can help see what happens.

ha0lyu, 2024-11-18:
Thanks for your reply. 
I'm not sure if there is a problem with DiffTest, **I want to confirm there is no problem in DiffTest**, please check the log file in the [zip file](https://github.com/user-attachments/files/17801204/basic_abort.zip) I attached before. I think I've attached the **_full message_**, if you think that's not it, please tell me **what the full message is**, and how to generate. 
BTW: I generate the log by run `emu -i /path2/test.bin 2>/dev/null 1>/path2/test.log`.

poemonsense, 2024-11-19:
Please also redirect stderr to file and provide us with full messages.

ha0lyu, 2024-11-19:
Run `./build/emu -i /home/path2/b1.bin > /home/path2/b1.log 2>&1`, got this: 
[b1.log](https://github.com/user-attachments/files/17814817/b1.log)
I have no more messages.

poemonsense, 2024-11-20:
In your log file: ` t0 different at pc = 0x0080000010, right= 0x0000000000000000, wrong = 0x0000000000000050`.

Though this log seems different from the last zip file you provided, I believe DiffTest shows the correct error information for this case.

closed 2024-11-20 by ha0lyu