OpenXiangShan/difftest #597: Emu: conflicting declaration of C function ‘void flash_read(uint32_t, uint64_t*)’

trueif, 2025-03-12:
### Before start

- [x] I have read the [XiangShan Documents](https://xiangshan-doc.readthedocs.io/zh_CN/latest). 我已经阅读过香山文档。
- [x] I have searched the previous issues and did not find anything relevant. 我已经搜索过之前的 issue，并没有找到相关的。
- [x] I have searched the previous discussions and did not find anything relevant. 我已经搜索过之前的 discussions，并没有找到相关的。

### Describe you problem

I meet the following confliction during "make emu CONFIG=MinimalConfig EMU_THREADS=2 -j10":

make[3]: Entering directory '/home/shuaige/kmh/XiangShanEmu/build/emu-compile'
In file included from /home/shuaige/kmh/XiangShanEmu/difftest/src/test/csrc/verilator/emu.cpp:20:
/home/shuaige/kmh/XiangShanEmu/difftest/src/test/csrc/common/flash.h:34: error: conflicting declaration of C function ‘void flash_read(uint32_t, uint64_t*)’
   34 | extern "C" void flash_read(uint32_t addr, uint64_t *data);
      |
In file included from ./VSimTop___024unit.h:10,
                 from ./VSimTop__Syms.h:14,
                 from /home/shuaige/kmh/XiangShanEmu/difftest/src/test/csrc/verilator/emu.h:21,
                 from /home/shuaige/kmh/XiangShanEmu/difftest/src/test/csrc/verilator/emu.cpp:17:
./VSimTop__Dpi.h:23: note: previous declaration ‘void flash_read(int, long long int*)’
   23 |     extern void flash_read(int addr, long long* data);

It seems like the parameter type is not consistent between the two files:
difftest/src/test/csrc/common/flash.h
build/emu-compile/VSimTop___024unit.h

### What did you do before

make clean
rm -rf out
make emu CONFIG=MinimalConfig EMU_THREADS=2 -j10

### Environment

XiangShan branch: master
XiangShan commit id: 58a45450fd3814b68acc144d2d89682d13a83dd3
NEMU commit id: 49366fa402efdbeee305785e15ff21b8e0a660f6
SPIKE commit id: ??
Operating System: 6.8.0-52-generic https://github.com/OpenXiangShan/XiangShan/pull/53~22.04.1-Ubuntu
gcc version: gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
mill version: 0.12.3
java version: openjdk 11.0.26 2025-01-21

poemonsense, 2025-03-12:
Which verilator version are you using? We need this information to track verilator changes.

closed 2025-03-12 by poemonsense

trueif, 2025-03-12:
Verilator 4.038 2020-07-11 rev v4.036-114-g0cd4a57ad

poemonsense, 2025-03-12:
> Verilator 4.038 2020-07-11 rev v4.036-114-g0cd4a57ad

Please do use the up-to-date Verilator version as indicated in https://github.com/OpenXiangShan/xs-env/blob/master/install-verilator.sh