OpenXiangShan/difftest #606: Using DiffTest with Rocket-Chip as DUT

ha0lyu, 2025-04-13:
**Related component**:  build Rocket

**Describe the question**
Hi, thanks for your great work on DiffTest!

I noticed that there’s a [dev-difftest](https://github.com/OpenXiangShan/rocket-chip/tree/dev-difftest) branch in the Rocket-Chip repo. I’ve cloned that branch and run make init to initialize submodules, etc., but encountered a failure when running make emu.

I also have an environment where XiangShan is used as the DUT. In that setup, running make emu inside `/home/xs-env/XiangShan/rocket-chip` also fails.

**What you expect us (DiffTest developers) to do for you**
Would it be possible to provide a brief tutorial or notes on how to successfully build and run DiffTest with Rocket as the DUT?

**Additional context**
Could the issue be related to the Verilator version? I’m currently using Verilator 5.028, which works fine for XiangShan.

poemonsense, 2025-04-14:
What's the error you encoutered? Please provide the log for us to help you.

`make emu` should work to build an executable.

ZhongYic00, 2025-04-14:
@poemonsense  I'm working on this now. Although compilation succeeds, using spike as reference (seems NEMU doesn't support rocket) always result in abort.

- build with `make emu` (default nemu ref), run without explicit `--diff` works
- build with `make emu` (default nemu ref), run with explicit `--diff=xxx-nemu-so` fails due to segfault
- build with `make emu` (default nemu ref), run with explicit `--diff=xxx-spike-so` fails due to abort in `decode_vm_info`
- build with `make emu REF=Spike`, run without explicit `--diff` fails due to abort in `decode_vm_info`

Tracing the failure shows that `proc->state.prv` is uninitialized. But I'm stuck here.

---

build env is configured according to CHIPYARD's official doc

poemonsense, 2025-04-14:
Have a try with the Spike in OpenXiangShan?

We maintain a Spike for cosim usage in OpenXiangShan organization. cd into the difftest directory and make CPU=ROCKET_CHIP should produce a correct so.

ZhongYic00, 2025-04-15:
Yeah, that's the one I tried

poemonsense, 2025-04-15:
These are my commands:

```
10011  git clone -b dev-difftest https://github.com/OpenXiangShan/rocket-chip
10012  git clone git@github.com:OpenXiangShan/riscv-isa-sim.git
10013  export NOOP_HOME=$(pwd)/rocket-chip
10014  export SPIKE_HOME=$(pwd)/riscv-isa-sim
10021  cd riscv-isa-sim/difftest
10022  make CPU=ROCKET_CHIP -j16
10023  cd ../../rocket-chip
10025  make init
10026  make emu -j16 REF=Spike
10027  ./build/emu -i ./common.sc
10028  ./build/emu -i ../../rocket/rocket-chip/_10018_
```

Should work perfectly. I just tried 2 minutes ago.

poemonsense, 2025-04-15:
> [@poemonsense](https://github.com/poemonsense) I'm working on this now. Although compilation succeeds, using spike as reference (seems NEMU doesn't support rocket) always result in abort.
>
> * build [...]

Most likely that your Spike is incorrectly built. Maybe you should get a clean machine, simply clone Spike, and build it. Do not use the `[sudo] make install` command listed in original readme to avoid destrorying your local environment.

ha0lyu, 2025-04-15:
> These are my commands:
>
> ```
> 10011  git clone -b dev-difftest https://github.com/OpenXiangShan/rocket-chip
> 10012  git clone git@github.com:OpenXiangShan/riscv-isa-sim.git
> 10013  export [...]

Thanks for your commands, I followed it and build difftest with rocket successfully.

poemonsense, 2025-04-15:
Thanks for reporting this

closed 2025-04-15 by poemonsense